rssnsj/openwrt-hc5x61 #5: 没有 kmod-iptunnel kmod-iptunnel4

matthew728960, 2015-01-11:
make menuconfig后输入“／”，就只有搜索到 kmod-iptunnel6 ，
kmod-iptunnel kmod-iptunnel4都是 unknown.

kmod-yavlan 则是 No matches found

版本：barrier_breaker 43929

MATTHEW.

rssnsj, 2015-01-11:
没有吗？有吧？
root@Large-Pipe:~# opkg list-installed | grep iptunnel
kmod-iptunnel - 3.10.49-1
kmod-iptunnel4 - 3.10.49-1
kmod-iptunnel6 - 3.10.49-1
root@Large-Pipe:~# opkg list-installed | grep yavlan
kmod-yavlan - 3.10.49-20141120
root@Large-Pipe:~#

rssnsj, 2015-01-11:
rssnsj@precise-aaa:~/openwrt-hc5761$ cat config-hiwifi-hc5761 | grep iptunnel
CONFIG_PACKAGE_kmod-iptunnel=y
CONFIG_PACKAGE_kmod-iptunnel4=y
CONFIG_PACKAGE_kmod-iptunnel6=y
rssnsj@precise-aaa:~/openwrt-hc5761$ cat config-hiwifi-hc5761 | grep yavlan
CONFIG_PACKAGE_kmod-yavlan=y
rssnsj@precise-aaa:~/openwrt-hc5761$

matthew728960, 2015-01-18:
都是我的错，我看错了，Sorry！

closed 2015-01-23 by rssnsj