rssnsj/openwrt-hc5x61 #13: HC5761的USB能够使用么？

paradislover, 2015-02-10:
发现插入优盘或者FTDI设备，一点反应都没有？需要安装哪些模块？

paradislover, 2015-02-10:
1. U盘插入，重启路由器可以识别，热插拔不能识别
2. FTDI设备不能识别

请问和DTS文件配置有关么？

paradislover, 2015-02-10:
附上插入FTDI设备的LOG：
[    0.000000] Board has DDR2
[    0.000000] Analog PMU set to hw control
[    0.000000] Digital PMU set to hw control
[    0.000000] SoC Type: Ralink MT7620A ver:2 eco:6
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU revision is: 00019650 (MIPS 24KEc)
[    0.000000] MIPS: machine is HiWiFi HC5761
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 08000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x00000000-0x07ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x00000000-0x07ffffff]
[    0.000000] On node 0 totalpages: 32768
[    0.000000] free_area_init_node: node 0, pgdat 802d4990, node_mem_map 81006600
[    0.000000]   Normal zone: 256 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 32512
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] PID hash table entries: 512 (order: -1, 2048 bytes)
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Writing ErrCtl register=0003b42a
[    0.000000] Readback ErrCtl register=0003b42a
[    0.000000] Memory: 126484k/131072k available (2396k kernel code, 4588k reserved, 592k data, 208k init, 0k highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:256
[    0.000000] CPU Clock: 580MHz
[    0.050000] Calibrating delay loop... 385.84 BogoMIPS (lpj=1929216)
[    0.060000] pid_max: default: 32768 minimum: 301
[    0.060000] Mount-cache hash table entries: 512
[    0.070000] pinctrl core: initialized pinctrl subsystem
[    0.070000] NET: Registered protocol family 16
[    0.080000] rt2880-pinmux pinctrl.1: try to register 73 pins ...
[    0.080000] pinctrl core: registered pin 0 (io0) on rt2880-pinmux
[    0.080000] pinctrl core: registered pin 1 (io1) on rt2880-pinmux
[    0.080000] pinctrl core: registered pin 2 (io2) on rt2880-pinmux
[    0.080000] pinctrl core: registered pin 3 (io3) on rt2880-pinmux
[    0.080000] pinctrl core: registered pin 4 (io4) on rt2880-pinmux
[    0.080000] pinctrl core: registered pin 5 (io5) on rt2880-pinmux
[    0.080000] pinctrl core: registered pin 6 (io6) on rt2880-pinmux
[    0.080000] pinctrl core: registered pin 7 (io7) on rt2880-pinmux
[    0.080000] pinctrl core: registered pin 8 (io8) on rt2880-pinmux
[    0.080000] pinctrl core: registered pin 9 (io9) on rt2880-pinmux
[    0.080000] pinctrl core: registered pin 10 (io10) on rt2880-pinmux
[    0.080000] pinctrl core: registered pin 11 (io11) on rt2880-pinmux
[    0.080000] pinctrl core: registered pin 12 (io12) on rt2880-pinmux
[    0.080000] pinctrl core: registered pin 13 (io13) on rt2880-pinmux
[    0.080000] pinctrl core: registered pin 14 (io14) on rt2880-pinmux
[    0.080000] pinctrl core: registered pin 15 (io15) on rt2880-pinmux
[    0.080000] pinctrl core: registered pin 16 (io16) on rt2880-pinmux
[    0.080000] pinctrl core: registered pin 17 (io17) on rt2880-pinmux
[    0.080000] pinctrl core: registered pin 18 (io18) on rt2880-pinmux
[    0.080000] pinctrl core: registered pin 19 (io19) on rt2880-pinmux
[    0.080000] pinctrl core: registered pin 20 (io20) on rt2880-pinmux
[    0.080000] pinctrl core: registered pin 21 (io21) on rt2880-pinmux
[    0.080000] pinctrl core: registered pin 22 (io22) on rt2880-pinmux
[    0.080000] pinctrl core: registered pin 23 (io23) on rt2880-pinmux
[    0.080000] pinctrl core: registered pin 24 (io24) on rt2880-pinmux
[    0.080000] pinctrl core: registered pin 25 (io25) on rt2880-pinmux
[    0.080000] pinctrl core: registered pin 26 (io26) on rt2880-pinmux
[    0.080000] pinctrl core: registered pin 27 (io27) on rt2880-pinmux
[... 294 lines cut ...]
[   23.420000] 8021q: adding VLAN 0 to HW filter on device eth0
[   23.440000] device eth0.1 entered promiscuous mode
[   23.440000] device eth0 entered promiscuous mode
[   23.460000] br-lan: port 1(eth0.1) entered forwarding state
[   23.470000] br-lan: port 1(eth0.1) entered forwarding state
[   24.210000] cfg80211: Calling CRDA for country: US
[   24.250000] cfg80211: Regulatory domain changed to country: US
[   24.250000] cfg80211:  DFS Master region: FCC
[   24.260000] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[   24.270000] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 3000 mBm), (N/A)
[   24.270000] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz), (N/A, 1700 mBm), (N/A)
[   24.280000] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz), (N/A, 2300 mBm), (0 s)
[   24.290000] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 3000 mBm), (N/A)
[   24.300000] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)
[   24.410000] ralink_soc_eth 10100000.ethernet eth0: port 5 link up (10Mbps/Half duplex)
[... cut ...]
[   26.480000] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   26.480000] device wlan0 entered promiscuous mode
[   26.490000] br-lan: port 2(wlan0) entered forwarding state
[   26.500000] br-lan: port 2(wlan0) entered forwarding state
[   26.500000] br-lan: port 2(wlan0) entered disabled state
[   34.960000] br-lan: port 2(wlan0) entered forwarding state
[   34.960000] br-lan: port 2(wlan0) entered forwarding state
[   34.970000] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   36.960000] br-lan: port 2(wlan0) entered forwarding state

rssnsj, 2015-02-10:
USB驱动已经完善，极2 USB能用。

paradislover, 2015-02-10:
多谢，已解决，编译时：
CONFIG_PACKAGE_kmod-usb-ohci=y 即可
==================================
MARK

closed 2015-09-14 by rssnsj