rssnsj/openwrt-hc5x61 #43: How to upgrade from 5681's stock firmware to openwrt?

hackruu, 2015-06-09:
I'm trying to update via tftp, but no luck.

```
TFTP from server 192.168.1.88; our IP address is 192.168.1.119
Filename 'recovery.bin'.

 TIMEOUT_COUNT=10,Load address: 0x80100000
Loading: Got ARP REPLY, set server/gtwy eth addr (b8:8d:12:53:98:68)
Got it
#################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 ##############
done
Bytes transferred = 8388612 (800004 hex)
NetBootFileXferSize= 00800004
recovery img chk failed!
```

hackruu, 2015-06-09:
when i'm trying to install rooted firmware i get:
```
TFTP from server 192.168.1.88; our IP address is 192.168.1.119
Filename 'recovery.bin'.

 TIMEOUT_COUNT=10,Load address: 0x80100000
Loading: Got ARP REPLY, set server/gtwy eth addr (b8:8d:12:53:98:68)
Got it
#################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 ############
done
Bytes transferred = 13369348 (cc0004 hex)
NetBootFileXferSize= 00cc0004
HiWiFi verify failed```

madwyn, 2015-06-09:
I hope you can understand Chinese. You can read the description of this project, then you will understand. Otherwise, please read on.

Please check your ROM version first. If it's above 9003, then it's impossible to flash OpenWrt directly.

That is because the bootloader is locked for third-party firmwares. You will have to gain root privilege first.

To do this, you'll need a functional China mainland mobile number and so on. Please just follow the [official guide](https://bbs.hiwifi.com/thread-74899-1-1.html).

hackruu, 2015-06-10:
thanks. i contacted hiwifi support via WeChat and they enabled developer mode for my hiwifi account. It was not easy, but i managed :)

madwyn, 2015-06-10:
I'm glad that helped. This issues could be closed.

closed 2015-06-10 by hackruu

wall25, 2015-12-23:
I bought this router, contact support, but they turned me down in unlocking developer mode. They said that developers can act only in China. :(
Is there another way to flash openwrt?