rssnsj/openwrt-hc5x61 #50: 建议：把极路由加入openwrt官方代码里。小米路由已经加入了……

sinkcup, 2015-07-09:
刚clone了openwrt Chaos Calmer (trunk)官方代码：https://dev.openwrt.org/wiki/GetSource
`
git clone git@git.coding.net:openwrtio/openwrt.git
git clone git://git.openwrt.org/openwrt.git
cd openwrt
make menuconfig
`
选了Ralink和MT7620，Target Profile里出现了小米路由 Xiaomi MiWiFi Mini，如图：
[image]
[image]

建议：把极路由也提交到openwrt官方代码里，便于开发者编译。

madwyn, 2015-07-15:
赞同，贡献到主干，少好多麻烦

OrangeYoung, 2015-07-21:
这个必须赞同

yasteora, 2015-08-05:
赞同，但是不知道 5Gwifi 驱动非开源是否会影响？

rssnsj, 2015-08-22:
确实很想，我也向openwrt官方提交过，不过没有精力跟进patch的后续进展，哪位兄弟能帮这个忙？

closed 2015-08-22 by rssnsj

sinkcup, 2015-09-05:
嗯。最近没怎么折腾路由器了。有空时研究下怎么提交到openwrt官方。

各位童鞋有空都研究研究啊。