rssnsj/openwrt-hc5x61 #55: 建议shadowsocks增加使用配置文件选项，以及非中国IP的代理模式中增加哪些IP不通过代理选项。

cielpy, 2015-07-28:
有时需要切换服务器，如果是使用配置文件的话会方便好多，另外代理所以非中国IP的模式下，有些IP想排除，但是似乎是没有选项的。。希望可以考虑下我的建议，感谢作者的付出:)

yasteora, 2015-08-26:
有个gfwlist 里面 可以自己 配置。不过我没用通过dnsmaqs 的ipset功能分组 ip 后通过ip 规则打标记 到table  在通过防火墙规则 来透明代理的方式，我直接全局代理。
ipset list 那里 似乎就是gfwlist之类的，我已经把那些个插件卸载了 帮不了你了，你自己看看  github有一本 叫做freerouter 的PDF 你可以看看

cielpy, 2015-08-26:
@yasteora 找到方法了，我用的是分国内外IP的模式，不是GFWList模式，只有几个IP不想走代理，所以把那几个IP加到ipset的国内列表里就行了。就是有Luci界面没有相关设置，自己手动添加了，不过也不麻烦。已经解决了，谢谢。

closed 2015-09-14 by rssnsj

Leeiio, 2015-12-07:
@cielpy 请问怎么修改？应该改哪个文件

cielpy, 2015-12-07:
/etc/ipset/local 里加上你不想走代理的ip就行了。

Leeiio, 2015-12-07:
@cielpy 看到了，多谢，不过我选国内外ip模式twitter、facebook这些还是会打不开..你有遇到这个问题么

cielpy, 2015-12-07:
如果你没改什么的话，设置好ss账号后就应该没问题了。DNS问题也是newwork-feed里解决了，看下你账号有没有填错。

Leeiio, 2015-12-07:
@cielpy 使用shadowsocks的简单模式设置ss是可以上网的，但是使用「面向高级用户」里的「所有非中国的IP」就会有这样的问题，可能是我杭州这的DNS污染太严重了...

cielpy, 2015-12-07:
Twitter等网站的DNS是不走本地的DNS解析的，用的是pdnsd转发到8.8.8.8，你可以ssh登录路由器，然后去年dig @127.0.0.1 -p 7453 twitter.com看返回什么结果

Leeiio, 2015-12-07:
刚才切回gfwlist模式可以返回结果，然后切回国内ip模式，就变成
root@OpenWrt:~# dig @127.0.0.1 -p 7453 twitter.com

; <<>> DiG 9.9.4 <<>> @127.0.0.1 -p 7453 twitter.com
; (1 server found)
;; global options: +cmd
;; connection timed out; no servers could be reached

Leeiio, 2015-12-07:
以下是gfwlist模式的返回结果

dig @127.0.0.1 -p 7453 twitter.com

; <<>> DiG 9.9.4 <<>> @127.0.0.1 -p 7453 twitter.com
; (1 server found)
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 28053
;; flags: qr rd ra; QUERY: 1, ANSWER: 2, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 1024
;; QUESTION SECTION:
;twitter.com.			IN	A

;; ANSWER SECTION:
twitter.com.		900	IN	A	104.244.42.193
twitter.com.		900	IN	A	104.244.42.1

;; Query time: 164 msec
;; SERVER: 127.0.0.1#7453(127.0.0.1)
;; WHEN: Mon Dec 07 15:43:29 GMT 2015
;; MSG SIZE  rcvd: 72

cielpy, 2015-12-07:
国内外模式域名不能解析，先用gfw模式吧，明天我看看国内外分流模式的dns怎么搞的。你可以先看看国内外ip分流模式下pdnsd有没有运行，试着把8.8.8.8加到ipset的过滤列表，看看会不会能解析。

Sent using CloudMagic [https://cloudmagic.com/k/d/mailapp?ct=pi&cv=7.1.27&pv=8.4]

[https://github.com/rssnsj/openwrt-hc5x61/issues/55#issuecomment-162561958] .[https://github.com/notifications/beacon/AC_jvRdMFnOpMJHJ9UXCnThIdlnvDv-qks5pNaBngaJpZM4Fg55Y.gif]

Leeiio, 2015-12-07:
pdnsd还真没接触过，以前用的是openwrt官网下的版本，然后自己装的shadowsocks和chinadns来翻墙的，但是最近十分不稳定，经常无法正常翻墙才换了这个固件...

cielpy, 2015-12-08:
ps看一下有没有pdnsd进城就行了

Sent using CloudMagic [https://cloudmagic.com/k/d/mailapp?ct=pi&cv=7.1.27&pv=8.4]

[https://github.com/rssnsj/openwrt-hc5x61/issues/55#issuecomment-162583205] .[https://github.com/notifications/beacon/AC_jvTvawgk00uMH9hjPufBjDjg6mXiIks5pNa05gaJpZM4Fg55Y.gif]

Leeiio, 2015-12-08:
果然，切到国内ip模式，pdnsd进城就没有了...

cielpy, 2015-12-08:
> 答复:  rssnsj/openwrt-hc5x61 <reply@reply.github.com>
> 日期:  2015年12月8日 星期二 上午9:51
> 至:  rssnsj/openwrt-hc5x61 <openwrt-hc5x61@noreply.github.com>

安全DNS打开了吗？

Leeiio, 2015-12-08:
@cielpy 安全DNS填写的是8.8.8.8，打开了没怎么看？

cielpy, 2015-12-08:
看代码是如果开了安全DNS并且DNS uses TCP勾上就会启动pdnsd的，并生成一个dnsmasq的域名解析列表，你看下有一个tcp dns之类的有没有勾上。

日期:  2015年12月8日 星期二 上午10:29

Leeiio, 2015-12-08:
@cielpy 好吧，原来必须要勾选「DNS使用TCP」才可以正常启动pdnsd的...这下似乎好了

cielpy, 2015-12-08:
嗯，代码逻辑是这样的，那个选项不是默认勾上的吗？

日期:  2015年12月8日 星期二 上午11:13

Leeiio, 2015-12-08:
@cielpy 并没有默认勾上，我刷的是最新的release固件，不是自己打补丁的
多谢了，最后问一下，这个固件里的ipset china文件里的ip不会自动更新的是吧？

cielpy, 2015-12-08:
不会。用这个命令更新吧。
wget -O- 'http://ftp.apnic.net/apnic/stats/apnic/delegated-apnic-latest' | awk -F\| '/CN\|ipv4/ { printf("%s/%d\n", $4, 32-log($5)/log(2)) }' > /etc/chinadns_chnroute.txt

最后那里是这路径，你改成你要写入的文件的路径就好，或者下载完了cp过去。

日期:  2015年12月8日 星期二 上午11:33

Leeiio, 2015-12-08:
不是应该是
```
curl 'http://ftp.apnic.net/apnic/stats/apnic/delegated-apnic-latest' | grep ipv4 | grep CN | awk -F\| '{ printf("add china %s/%d\n", $4, 32-log($5)/log(2)) }' > ~/Desktop/chinadns_chnroute.txt
``` 
这样么,我看china文件里的每行ip前缀是add china，`add china 1.0.1.0/24`，而且第一行是写着`create china hash:net family inet hashsize 1024 maxelem 65536`，没有`#`注释，不知道是不是也需要的

cielpy, 2015-12-08:
0.0，是，用你那个命令就行了。不用每天 更新，这个列表很少变化，隔段时间更新一下就行了。

日期:  2015年12月8日 星期二 上午11:40

cielpy, 2015-12-08:
第一行那个也是需要的，更新过IP列表后要加上。创建的ipset标记。

Leeiio, 2015-12-08:
第一行内容不知道要怎么自动添加到文件第一行...

cielpy, 2015-12-08:
手动吧。

日期:  2015年12月8日 星期二 下午2:07

Leeiio, 2015-12-08:
研究了下awk命令发现这样写就可以了
```
curl 'http://ftp.apnic.net/apnic/stats/apnic/delegated-apnic-latest' | grep ipv4 | grep CN | awk -F\| 'BEGIN{printf "create china hash:net family inet hashsize 1024 maxelem 65536\n"}{printf("add china %s/%d\n", $4, 32-log($5)/log(2)) }' > /etc/ipset/china
```
写入计划任务
```
# 自动更新
# 每天凌晨4点更新文件，注意文件路径
* 4 * * * curl 'http://ftp.apnic.net/apnic/stats/apnic/delegated-apnic-latest' | grep ipv4 | grep CN | awk -F\| 'BEGIN{printf "create china hash:net family inet hashsize 1024 maxelem 65536\n"}{printf("add china %s/%d\n", $4, 32-log($5)/log(2)) }' > /etc/ipset/china
```

cielpy, 2015-12-08:
BEGIN是执行一次吗。

日期:  2015年12月8日 星期二 下午2:53

Leeiio, 2015-12-08:
是的，亲测可用

cielpy, 2015-12-08:
嗯，我现在不用这个了，我把这些卸载掉了，直接装了官方的shadowsocks，官方的包会处理这些ipset，只需要指定一个IP列表就行了。

日期:  2015年12月8日 星期二 下午2:57

Leeiio, 2015-12-08:
shadowsocks-libev-spec 还是shadowsocks-libev？
但是还是要额外处理DNS污染问题吧 我之前也是用官方的shadowsocks配合的chinadns

cielpy, 2015-12-08:
shadowsocks-libev-spec只用ss-redir提供透明代理，DNS污染问题参考了这个项目。
这个项目的DNS是用pdnsd通过tcponly的方案请求8.8.8.8，前两天我这用这个方案得不到解析结果，我所以我换成了shadowsocks-libev+dns2socks+pdnsd的方式
跑一个ss-local，提供给dns2socks一个socks5代理，然后把pdnsd的上游设置成了dns2socks提供缓存。用越来还行。

日期:  2015年12月8日 星期二 下午3:04

Leeiio, 2015-12-09:
你有用15.05么？我发现里面自带去掉了pdnsd，也不知道15.05是否更稳定

cielpy, 2015-12-09:
是的，用的15.05，装的14.07里的那个pdnsd，没发现什么问题。

Leeiio, 2015-12-10:
可否分享下你的pdnsd配置？

cielpy, 2015-12-10:
global {
perm_cache=2048;
cache_dir="/var/pdnsd";
pid_file = /var/run/pdnsd.pid;
run_as="nobody";
server_ip = 127.0.0.1;
server_port = 7453;
status_ctl = on;
query_method = tcp_only;
min_ttl=15m;
max_ttl=1w;
timeout=10;
neg_domain_pol=on;
proc_limit=2;
procq_limit=8;
}
server {
label= "dns";
ip = 8.8.8.8;
timeout=6;
uptest=none;
interval=10m;
purge_cache=off;
}

日期:  2015年12月10日 星期四 上午11:18

Leeiio, 2015-12-10:
这样配置似乎没法指定国内的ip走国内的dns了是不？
为了防止国内的isp劫持需要用223.5.5.5这样的dns...

cielpy, 2015-12-10:
这个只是提供一个国外的纯净DNS，可以另外配合了dnsmasq让国内常用域名走国内DNS
用这个项目https://github.com/felixonmars/dnsmasq-china-list
把国内的常用域名走自定义的国内DNS，其他的全部走了pdnsd。

日期:  2015年12月10日 星期四 上午11:58

Leeiio, 2015-12-10:
不喜欢基于域名的方案，维护的列表太多了而且是一个不可控的不断增加
global {
perm_cache=2048;
cache_dir="/var/pdnsd";
pid_file = /var/run/pdnsd.pid;
run_as="nobody";
server_ip = 127.0.0.1;
server_port = 7453;
status_ctl = on;
query_method = tcp_only;
min_ttl=15m;
max_ttl=1w;
timeout=10;
neg_domain_pol=on;
proc_limit=2;
procq_limit=8;
}
server {
        label = "china DNS";
        ip = 223.5.5.5,
             119.29.29.29;
        timeout = 3;
}
server {
label= "dns";
ip = 8.8.8.8;
timeout=6;
uptest=none;
interval=10m;
purge_cache=off;
}
不知道是否可以这样

cielpy, 2015-12-10:
没办法，但是现在没有自动化方案，ChinaDNS算比较自动，但是自动化往往更不可控，CDN什么的，自动处理并不简单。

日期:  2015年12月10日 星期四 下午12:04