rssnsj/openwrt-hc5x61 #61: 刷好之后搜不到5G网络

stcatz, 2015-08-31:
5G网络需要额外配置什么吗？ 

路由器上的5G灯一直亮着。信息如下

信号: 0 dBm / 噪声: 0 dBm
0%	模式: Client | SSID: mao-5G
BSSID: 00:00:00:00:00:20 | 加密: -
信道: 1 (0.000 GHz) | 传输功率: 0 dBm
信号: 0 dBm | 噪声: 0 dBm
传输速率: 300.0 Mbit/s | 国家:00

stcatz, 2015-08-31:
我的 /etc/config/wireless

config wifi-device 'radio0'
        option type 'mac80211'
        option channel '11'
        option hwmode '11g'
        option path '10180000.wmac'
        option htmode 'HT20'
        option txpower '20'
        option country '00'

config wifi-iface
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'mao'
        option encryption 'psk2'
        option key '13830087489'

config wifi-device 'mt7612e'
        option type 'mt7612e'
        option vendor 'ralink'
        option band '5G'
        option channel '0'
        option autoch '2'
        option wifimode '9'
        option bw '1'
        option country 'None'
        option region '5'
        option bgprotect '0'
        option beacon '100'
        option dtim '1'
        option fragthres '2346'
        option rtsthres '2347'
        option txpower '100'
        option txpreamble '0'
        option shortslot '1'
        option txburst '1'
        option pktaggre '1'
        option ieee80211h '0'
        option ht_bsscoexist '0'
        option ht_extcha '1'
        option ht_opmode '0'
        option ht_gi '1'
        option ht_rdg '1'
        option ht_stbc '1'
        option ht_amsdu '0'
        option ht_autoba '1'
        option ht_badec '0'
        option ht_distkip '1'
        option ht_ldpc '0'
        option ht_txstream '2'
        option ht_rxstream '2'
        option radio '1'

config wifi-iface
        option device 'mt7612e'
        option ifname 'rai0'
        option network 'lan'
        option mode 'ap'
        option ssid 'mao-5G'
        option wmm '1'
        option apsd '0'
        option encryption 'psk2+ccmp'
        option key '13830087489'

rssnsj, 2015-09-14:
再确认一下，重启试试~

closed 2015-09-14 by rssnsj

js-john, 2017-03-07:
你好，我刚刷上openwrt也遇到了和你一样的问题，请问最后解决了吗？