rssnsj/openwrt-hc5x61 #81: 请教rssnjs，为什么编译成功完成了，却没有生成升级文件

jeffersonchou, 2015-11-01:
操作步骤如下：

1. Clone代码，并切换到chaos_calmer 分支
2. 复制config-hiwifi-hc5661a为config-hiwifi
3. 进入openwrt-hc5x61目录，make

但是./openwrt-ramips/bin/ 目录无生成openwrt-ramips-mt7620a-hiwifi-hc5661a-squashfs-sysupgrade.bin
文件，其他文件都是齐；make后控制台输出如下：
```
...
Generating index for package ./luci-lib-nixio_git-15.301.40538-a66c088-1_ramips_24kec.ipk
Generating index for package ./luci-mod-admin-full_git-15.301.40538-a66c088-1_ramips_24kec.ipk
Generating index for package ./luci-proto-ipv6_git-15.301.40538-a66c088-1_all.ipk
Generating index for package ./luci-proto-ppp_git-15.301.40538-a66c088-1_all.ipk
Generating index for package ./luci-theme-bootstrap_git-15.301.40538-a66c088-1_all.ipk
Generating index for package ./luci_git-15.301.40538-a66c088-1_all.ipk
Signing package index...
make[3]: Leaving directory `/home/jeff/openwrt-hc5661a/openwrt-ramips'
make[2]: Leaving directory `/home/jeff/openwrt-hc5661a/openwrt-ramips'
make[1]: Leaving directory `/home/jeff/openwrt-hc5661a/openwrt-ramips'
```
能帮忙看一下吗？
谢谢！

jeffersonchou, 2015-11-01:
编译平台是ubuntu14.04

ZhaoxiongTan, 2016-01-11:
同样遇到此问题