rssnsj/openwrt-hc5x61 #100: HiWiFi HC5861. How to enable developer mode without going to the official technical support?

wall25, 2015-12-23:
I bought this router (HiWiFi HC5861), contact support, but they turned me down in unlocking developer mode. They said that developers can act only in China. :(
Is there another way to flash openwrt?

tatsuteng, 2016-01-23:
You can hack the '''Aira2 installed in the SD card''' to gain root access and flash openwrt. FYI: http://hiwifiroot.lofter.com/post/1d0319c1_6596c99

madjapahit, 2017-11-18:
hi tatsuteng, the link has gone, can you add another link as reference? thanks!

tatsuteng, 2018-01-14:
I find another here: http://roothiwifi.lofter.com/post/1d733833_8685fcd
Here's the brief process if you have difficulties in reading Chinese:
1. Install a HiWiFi plugin "搜狐视频远程下载" that contains `aria2c`, which auto starts with root privilege on boot.
2. Because `aria2` was installed to a SD card, we can modify the `aria2c` file to execute a `dropbear` command
3. A SSH root access gained.

Alternatively, you can just use TTL to get a root shell.