rssnsj/openwrt-hc5x61 #110: svn 服务器问题

sheepbao, 2016-03-14:
您好，谢谢您rssnsj：
make 后会出现svn的服务无法访问：
`svn co svn://svn.openwrt.org/openwrt/branches/chaos_calmer openwrt-ramips -r46893
svn: E000111: Unable to connect to a repository at URL 'svn://svn.openwrt.org/openwrt/branches/chaos_calmer'
svn: E000111: Can't connect to host 'svn.openwrt.org': Connection refused
Makefile:52: recipe for target '.checkout_svn' failed
`
我已翻墙，防火墙也关闭了，也ping的通
`ping svn.openwrt.org
PING svn.openwrt.org (78.24.191.177) 56(84) bytes of data.
64 bytes from openwrt.org (78.24.191.177): icmp_seq=1 ttl=45 time=730 ms
64 bytes from openwrt.org (78.24.191.177): icmp_seq=2 ttl=45 time=370 ms
64 bytes from openwrt.org (78.24.191.177): icmp_seq=3 ttl=45 time=388 ms
64 bytes from openwrt.org (78.24.191.177): icmp_seq=4 ttl=45 time=385 ms
64 bytes from openwrt.org (78.24.191.177): icmp_seq=5 ttl=45 time=389 ms
64 bytes from openwrt.org (78.24.191.177): icmp_seq=6 ttl=45 time=389 ms
64 bytes from openwrt.org (78.24.191.177): icmp_seq=7 ttl=45 time=382 ms
`
这是什么原因？

zhanghanbin, 2016-03-26:
把Makefile里面的svn co svn://svn.openwrt.org/openwrt/branches/barrier_breaker $(openwrt_dir) -r43770改成svn co svn://svn.openwrt.org.cn/openwrt/branches/chaos_calmer $(openwrt_dir)

sheepbao, 2016-04-12:
谢谢，可以改成用git服务吗？在国内也比较好下载