rssnsj/openwrt-hc5x61 #115: 求助：IPv6 ping不通

MrJasper, 2016-04-10:
```
config dhcp wan6
    # 添加下面这一行，我的环境没这行不成，有可路由的合法v6地址但是没有转发到wan口，添加了就好了
    option interface 'wan'
    option dhcpv6 relay
    option ra relay
    option ndp relay
    option master 1

config dhcp lan
    option dhcpv6 relay
    option ra relay
    option ndp relay
```
使用上面的配置，现在局域网内的设备也可以获取IPv6的地址：
[image]
但是ping命令显示：
```
正在 Ping orange.kame.net [2001:200:dff:fff1:216:3eff:feb1:44d7] 具有 32 字节的数据
无法访问目标网。
无法访问目标网。
无法访问目标网。
无法访问目标网。

2001:200:dff:fff1:216:3eff:feb1:44d7 的 Ping 统计信息:
    数据包: 已发送 = 4，已接收 = 0，丢失 = 4 (100% 丢失)，
```
请问要怎么配置来解决这个问题？

closed 2016-11-06 by MrJasper

TurnOffNOD, 2018-07-05:
@Jasper0819X ，这个问题最后解决了没？

Jasper0819X, 2018-07-05:
@TurnOffNOD 
没有，后来改用NAT6了，我学校的网络环境不支持relay