rssnsj/openwrt-hc5x61 #123: pdnsd源码获取不到

ukid, 2016-07-12:
`mkdir -p /data/openwrt-hc5x61/openwrt-ramips/dl`
`echo "Checking out files from the git repository..."; mkdir -p /data/openwrt-hc5x61/openwrt-ramips/tmp/dl && cd /data/openwrt-hc5x61/openwrt-ramips/tmp/dl && rm -rf pdnsd-1.2.9a-par && [ \! -d pdnsd-1.2.9a-par ] && git clone git://gitorious.org/pdnsd/pdnsd.git pdnsd-1.2.9a-par --recursive && (cd pdnsd-1.2.9a-par && git checkout a8e46ccba7b0fa2230d6c42ab6dcd92926f6c21d && git submodule update) && echo "Packing checkout..." && rm -rf pdnsd-1.2.9a-par/.git &&      /bin/tar czf /data/openwrt-hc5x61/openwrt-ramips/tmp/dl/pdnsd-1.2.9a-par-a8e46ccba7b0fa2230d6c42ab6dcd92926f6c21d.tar.gz pdnsd-1.2.9a-par && mv /data/openwrt-hc5x61/openwrt-ramips/tmp/dl/pdnsd-1.2.9a-par-a8e46ccba7b0fa2230d6c42ab6dcd92926f6c21d.tar.gz /data/openwrt-hc5x61/openwrt-ramips/dl/ && rm -rf pdnsd-1.2.9a-par;`
`Checking out files from the git repository...`
`Cloning into 'pdnsd-1.2.9a-par'...`
`fatal: unable to connect to gitorious.org:`
`gitorious.org[0: 64.13.172.37]: errno=Connection refused`
`gitorious.org[1: 2a02:c0:1014::1]: errno=Network is unreachable`
``
`make[4]: *** [/data/openwrt-hc5x61/openwrt-ramips/dl/pdnsd-1.2.9a-par-a8e46ccba7b0fa2230d6c42ab6dcd92926f6c21d.tar.gz] Error 128`
`make[4]: Leaving directory `/data/openwrt-hc5x61/openwrt-ramips/feeds/oldpackages/net/pdnsd'`
`make[3]: *** [package/feeds/oldpackages/pdnsd/compile] Error 2`
`make[3]: Leaving directory `/data/openwrt-hc5x61/openwrt-ramips'`
`make[2]: *** [/data/openwrt-hc5x61/openwrt-ramips/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/stamp/.package_compile] Error 2`
`make[2]: Leaving directory `/data/openwrt-hc5x61/openwrt-ramips'`
`make[1]: *** [world] Error 2`
`make[1]: Leaving directory `/data/openwrt-hc5x61/openwrt-ramips'`
`make: *** [HC5X61] Error 2`

ukid, 2016-07-19:
手动从https://gitorious.org/pdnsd/pdnsd.git处克隆下来可以解决。
makefile里面找不到，不之如何修改。

873314461, 2017-01-12:
克隆下来之后，放在哪里啊？
我也遇到了这个问题，pdnsd项目好像已经被删除了，我在github上找了一份
https://github.com/SAPikachu/pdnsd
但是不知道放在哪里

ukid, 2017-01-12:
````
echo "Checking out files from the git repository...";
mkdir -p /data/openwrt-hc5x61/openwrt-ramips/tmp/dl 
cd /data/openwrt-hc5x61/openwrt-ramips/tmp/dl 
rm -rf pdnsd-1.2.9a-par
[ \! -d pdnsd-1.2.9a-par ]
git clone git://gitorious.org/pdnsd/pdnsd.git pdnsd-1.2.9a-par --recursive
(cd pdnsd-1.2.9a-par && git checkout a8e46ccba7b0fa2230d6c42ab6dcd92926f6c21d && git submodule update)
echo "Packing checkout..." 
rm -rf pdnsd-1.2.9a-par/.git
/bin/tar czf /data/openwrt-hc5x61/openwrt-ramips/tmp/dl/pdnsd-1.2.9a-par-a8e46ccba7b0fa2230d6c42ab6dcd92926f6c21d.tar.gz pdnsd-1.2.9a-par
mv /data/openwrt-hc5x61/openwrt-ramips/tmp/dl/pdnsd-1.2.9a-par-a8e46ccba7b0fa2230d6c42ab6dcd92926f6c21d.tar.gz /data/openwrt-hc5x61/openwrt-ramips/dl/ 
rm -rf pdnsd-1.2.9a-par;
````
这里有路径

codingbelief, 2017-02-28:
遇到同样的问题，解决方式如下：
修改文件 openwrt-hc5x61/openwrt-ramips/feeds/oldpackages/net/pdnsd/Makefile 中

PKG_SOURCE_URL:=git://gitorious.org/pdnsd/pdnsd.git

为 

PKG_SOURCE_URL:=http://gitorious.org/pdnsd/pdnsd.git