rssnsj/openwrt-hc5x61 #130: 你好，我用的是极路由1S，7628那款芯片那款，用你的imagebuilder-MT7628的编译错误

fantasylidong, 2017-01-06:
在/root/openwrt-hc5x61-cc/ImageBuilder-MT7628文件夹运行这个命令
make BUILDER=OpenWrt-ImageBuilder-15.05-ramips-mt7628.Linux-x86_64 SDK=OpenWrt-SDK-15.05-ramips-mt7628_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64 HC5661A
运行log如下：
/bin/sh: 1: cd: can't cd to OpenWrt-SDK-15.05-ramips-mt7628_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/bin/ramips/packages
mkdir -p OpenWrt-ImageBuilder-15.05-ramips-mt7628.Linux-x86_64/dl
cp -f repo-base.conf OpenWrt-ImageBuilder-15.05-ramips-mt7628.Linux-x86_64/repositories.conf
mkdir -p OpenWrt-ImageBuilder-15.05-ramips-mt7628.Linux-x86_64/target/linux/ramips/base-files/etc
cp -f opkg.conf OpenWrt-ImageBuilder-15.05-ramips-mt7628.Linux-x86_64/target/linux/ramips/base-files/etc/opkg.conf
make -C OpenWrt-ImageBuilder-15.05-ramips-mt7628.Linux-x86_64 image PROFILE=HiWiFi-HC5661A \
	FILES="/root/openwrt-hc5x61-cc/ImageBuilder-MT7628/OpenWrt-ImageBuilder-15.05-ramips-mt7628.Linux-x86_64/target/linux/ramips/base-files" \
	PACKAGES="6in4 6to4 blkid curl ethtool fdisk ip ip6tables-extra ip6tables-mod-nat ipset iptables-mod-conntrack-extra iptables-mod-extra iptables-mod-ipopt iwinfo iptables-mod-nat-extra kmod-crypto-des kmod-crypto-ecb kmod-crypto-hmac kmod-crypto-manager kmod-crypto-md4 kmod-crypto-md5 kmod-crypto-pcompress kmod-crypto-sha1 kmod-crypto-sha256 kmod-dnsresolver kmod-fs-ext4 kmod-fs-vfat kmod-fs-cifs kmod-fuse kmod-gre kmod-ipt-filter kmod-ip6-tunnel kmod-ip6tables-extra kmod-ipip kmod-ipt-conntrack-extra kmod-ipt-extra kmod-ipt-ipopt kmod-ipt-ipset kmod-ipt-nat-extra kmod-ipt-nat6 kmod-iptunnel kmod-iptunnel4 kmod-iptunnel6 kmod-leds-gpio kmod-ledtrig-default-on kmod-ledtrig-netdev kmod-lib-zlib kmod-ledtrig-timer kmod-ledtrig-usbdev kmod-lib-textsearch kmod-macvlan kmod-mppe kmod-nfnetlink kmod-pptp kmod-sit kmod-tun kmod-usb-storage libblkid libcurl libdaemon libevent2 liblzo libmnl libncurses libopenssl libpolarssl libpthread luci-app-samba luci-lib-json luci-proto-ipv6 luci-proto-relay ntfs-3g openvpn-openssl resolveip samba36-server terminfo zlib luci-i18n-base-zh-cn luci-i18n-commands-zh-cn luci-i18n-diag-core-zh-cn luci-i18n-firewall-zh-cn luci-i18n-qos-zh-cn luci-i18n-samba-zh-cn ppp-mod-pptp ppp-mod-pppol2tp kmod-nls-cp437 kmod-nls-iso8859-1 kmod-nls-utf8 kmod-crypto-deflate kmod-usb-ohci kmod-usb-printer kmod-nf-nathelper kmod-nf-nathelper-extra "
make[1]: Entering directory '/root/openwrt-hc5x61-cc/ImageBuilder-MT7628/OpenWrt-ImageBuilder-15.05-ramips-mt7628.Linux-x86_64'
Profile "HiWiFi-HC5661A" does not exist!
Use "make info" to get a list of available profile names.
Makefile:173: recipe for target 'image' failed
make[1]: *** [image] Error 1
make[1]: Leaving directory '/root/openwrt-hc5x61-cc/ImageBuilder-MT7628/OpenWrt-ImageBuilder-15.05-ramips-mt7628.Linux-x86_64'
Makefile:56: recipe for target 'HC5661A' failed
make: *** [HC5661A] Error 2

fantasylidong, 2017-01-06:
不知道是哪一步弄错了，文件全部是在openwrt官网上下的。具体链接如下
https://downloads.openwrt.org/chaos_calmer/15.05/ramips/mt7628/

873314461, 2017-01-12:
你应该用14.07版本

fantasylidong, 2017-01-12:
https://downloads.openwrt.org/barrier_breaker/14.07/ramips/
里面没有Mt 7628这款芯片啊

873314461, 2017-01-12:
那你用chaos_calmer分支试试呢

fantasylidong, 2017-01-12:
那个不支持极路由1s,，只支持7620那款芯片的极路由1，不知道怎么修改才能适配极路由1s，有相关教程么，我知道这个问题有点小白[尴尬]

mineyosemite, 2017-01-20:
你好，请问你编译好了吗？我也是hc5661a

wc791775547, 2017-09-21:
请问，这个问题解决了吗？求指导

yujincheng08, 2017-09-22:
@wc791775547 直接下载lede的hc5661a版本就行。

iTofu, 2022-06-14:
https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=4044268&page=1#pid9795907
这个固件亲测可用，MT7628A CPU

closed 2022-06-21 by fantasylidong