rssnsj/openwrt-hc5x61 #147: 编译报错recipe for target '.patched' failed

fowse2020, 2019-12-28:
git clone https://github.com/rssnsj/openwrt-hc5x61.git
编译报错，输出如下：
``` bash
make
Cloning into 'openwrt-ramips'...
remote: Enumerating objects: 10587, done.
remote: Total 10587 (delta 0), reused 0 (delta 0), pack-reused 10587
Receiving objects: 100% (10587/10587), 12.93 MiB | 484.00 KiB/s, done.
Resolving deltas: 100% (3698/3698), done.
HEAD is now at b763ba2 BB: openssl: update to 1.0.2f (fixes CVE-2016-0701, CVE-2015-3197)
*** Please make a symbolic of either config file to 'config'.
Makefile:39: recipe for target '.patched' failed
make: *** [.patched] Error 1
```
看了下“config-hiwifi-hc5761”的配置，不知道要修改哪里。本机环境为Ubuntu 18.04 LTS，make编译后`openwrt-ramips`目录下并未生成`bin`目录。

lnzdl, 2021-08-01:
一样问题 未解决

JerryCheng1, 2021-08-15:
mv config-hiwifi-hc5761 config